ASNeG/OpcUaStack #11: Rework Thread concept

huebl, 2018-10-02:
The thread is currently single threaded. This is problematic in several ways.

- The processing of slow tasks blocks other tasks

- The application uses the same thread as the SDK.

- The processor is not used optimally

huebl, 2018-12-26:
In all Software components of the application a uniform thread concept should be used. This concept should support the following concepts.

- All operations should be performed by non-blocking tasks.
- All operations are inserted into a queue and executed by one or more threads.

The following operations are supported.

- Asynchronous tasks
- Timer tasks
- Socket Tasks
- Message Tasks

huebl, 2018-12-27:
If an application consists of several software components, there are often dependencies between the software components during startup. A software component can go through several init states during startup. These dependencies between the software components should be able to be set using the configuration file.

huebl, 2019-01-25:
With this issue, a concept should first be worked out.

huebl, 2019-02-01:
In the first step, only a documentation is to be created.

flipback, 2019-06-21:
> `The thread is currently single threaded. This is problematic in several 

Do you meen OPC UA Session instead of "thread"?

huebl, 2019-12-29:
This feature is to be implemented with Issue #339. For this reason the feature is closed.

closed 2019-12-29 by huebl