ASNeG/OpcUaStack #58: Transfer of the information model to the application during startup

huebl, 2018-12-16:
If a node in the information model is to be accessed, the current reference to the node must be determined via a request. This is expensive. Therefore, it makes sense to pass an interface for the information model to the application.

flipback, 2018-12-17:
Yes, the transaction model is expensive, but it give us more control and flexibility if we need to change something in the stack or add some features. For example, we could provide access not only to local information model, but to model of the other application or even other application server. 

I suggest discussing it in #59.

flipback, 2018-12-21:
Hi, @huebl. I suppose we fixed it in #64, didn't we?

huebl, 2018-12-21:
Yes, this bug is fixed

closed 2018-12-21 by huebl