ASNeG/OpcUaStack #87: Local discovery doesn't work

flipback, 2019-01-12:
I have built a new user app and launched it. Then I tried to find it by using the discovery and got an error:

```
13:52:30.050 | General            |                                | [uastack] OpcUa_SecureConnection_OnNotify: Connect event: ERROR 0x80050000!
13:52:30.050 | General            |                                | [uastack] OpcUa_SecureConnection_OnNotify: Connect event: Notifying owner! 0x80050000
13:52:30.050 | DiscoveryWidget    |                                | Discovery FindServersOnNetwork on opc.tcp://localhost:4840 failed (BadCommunicationError), falling back to FindServers
13:52:30.051 | General            |                                | [uastack] OpcUa_SecureConnection_OnNotify: Connect event: ERROR 0x80050000!
13:52:30.051 | General            |                                | [uastack] OpcUa_SecureConnection_OnNotify: Connect event: Notifying owner! 0x80050000
13:52:30.051 | DiscoveryWidget    |                                | Discovery FindServers on opc.tcp://localhost:4840 failed (BadCommunicationError)
13:52:30.051 | General            |                                | [uastack] OpcUa_SecureConnection_OnNotify: Connect event: ERROR 0x80050000!
13:52:30.051 | General            |                                | [uastack] OpcUa_SecureConnection_OnNotify: Connect event: Notifying owner! 0x80050000
13:52:30.051 | DiscoveryWidget    |                                | Discovery GetEndpoints on opc.tcp://localhost:4840 failed (BadCommunicationError)
```
I use:
Stack's version 3.6.1
UAExpert 1.5
Ubuntu 16.04

flipback, 2019-01-12:
It seems It's not a bug but a feature request. The stack doesn't provide a discovery server, only a client. Am I right, @huebl?

closed 2019-01-12 by flipback

huebl, 2019-01-12:
Hi @flipback,

OPC UA Distinguishes two different types of servers. The Application Server offers a service. The Application Service registers its IP address in a discoyery server. Clients have the possibility to find the service.

The Discovery Server is a separate server. This server has functions for registering services and finding services. The OPC UA stack contains these functions.

flipback, 2019-01-12:
Yes, I have already understood that I've used it in the wrong way.