ASNeG/OpcUaStack #109: Server crashes during restart

flipback, 2019-02-07:
If the server restart and there are connected clients, it crushes sometimes with the following message in the log:

```
19-02-06 15:52:37:381 INFO  [9240] [..\src\servermain.cpp:58]   [2418] accepted new secure channel from client : Address=<127.0.0.1>, Port=<50436>
2019-02-06 15:52:37:388 INFO  [9240] [..\src\servermain.cpp:58]   [2418] accepted new secure channel from client : Address=<127.0.0.1>, Port=<50437>
2019-02-06 15:52:37:394 ERROR [9240] [..\src\servermain.cpp:64]   [2418] receive invalid channel id in OpenSecureChannelRequest : Local-Address=<0.0.0.0>, Local-Port=<48012>, Partner-Address=<127.0.0.1>, Partner-Port=<50436>, ChannelId=<10>
2019-02-06 15:52:37:397 ERROR [9240] [..\src\servermain.cpp:64]   [2418] receive invalid channel id in OpenSecureChannelRequest : Local-Address=<0.0.0.0>, Local-Port=<48012>, Partner-Address=<127.0.0.1>, Partner-Port=<50437>, ChannelId=<12>
```

Restart sequence:

```cpp
    opc_server->stop();
    opc_server->shutdown();  

    LOG_INFO("Init Server IO");

    ConfigXml xmlParser;
    xmlParser.parse("OpcUaServer.xml", &conf);

    opc_server->config(conf);

    opc_server->init();
    opc_server->start();
```

where `opc_server` is `std::shared_ptr<OpcUaStackServer::Server>`

OS: Windows 7
Version: 3.6.2

huebl, 2019-02-24:
**Error Description:**

After restarting the application, the channel should be reopened by the client. The client wants to reuse the previous channel and sends a reconnect open secure channel request. The channel does not exist anymore. Therefore, the server closes the connection. Thereafter, the server accesses the closed socket again. 

**Error Fixing:**
After closing the socket it must not be used anymore,

huebl, 2019-03-15:
The error description described above has not been confirmed. Testing with multiple OPC UA clients (Prosys OPC UA Client and UAExpert) did not cause any crashes.

@flipback - Can you provide a CORE dump for this bug?

huebl, 2019-03-15:
It could be a problem specific to Windows.

```
  __declspec(deprecated("By default, this function always fails with "
        "operation_not_supported when used on Windows XP, Windows Server 2003, "
        "or earlier. Consult documentation for details."))
#endif
  void cancel()
  {
    boost::system::error_code ec;
    this->get_service().cancel(this->get_implementation(), ec);
    boost::asio::detail::throw_error(ec, "cancel");
  }
```

flipback, 2019-03-16:
> The cancel command should be replaced by the close command. Maybe that fixes the bug.

Sorry, can't understand where and what I should replace.

flipback, 2019-03-16:
@huebl , I see. If  you add `Closes #109` into related PR, I'll see in this issue that there is a solution. :wink:

huebl, 2019-03-16:
I hope so

flipback, 2019-03-16:
> I hope so

I mean PR's description on GitHub not the commit.

PR #143 (2019-03-18): Replace cancel command by close command.
Closes #109
closed 2019-03-18 by huebl