ASNeG/OpcUaStack #110: AttributeBase::setValue compilation problem

flipback, 2019-02-07:
After updating the stack to 3.6.2 version in our project we've got a very confusing link error:

```
alarm_opc_mapper.cpp.obj : error LNK2019: ������ �� ������������� ������� ������ "public: bool __cdecl OpcUaStackServer::AttributeBase::setValue(class OpcUaStackCore::OpcUaDataValue const &)" (?setValue@AttributeBase@OpcUaStackServer@@QEAA_NAEBVOpcUaDataValue@OpcUaStackCore@@@Z) � ������� "public: void __cdecl serverio::pylon::alarm::AlarmOpcMapper::UpdateConditions(struct serverio::pylon::alarm::AlarmInfo const &,class boost::shared_ptr<class OpcUaStackCore::DiscreteAlarmType>)const " (?UpdateConditions@AlarmOpcMapper@alarm@pylon@serverio@@QEBAXAEBUAlarmInfo@234@V?$shared_ptr@VDiscreteAlarmType@OpcUaStackCore@@@boost@@@Z)
opc_information_model.cpp.obj : error LNK2001: ������������� ������� ������ ""public: bool __cdecl OpcUaStackServer::AttributeBase::setValue(class OpcUaStackCore::OpcUaDataValue const &)" (?setValue@AttributeBase@OpcUaStackServer@@QEAA_NAEBVOpcUaDataValue@OpcUaStackCore@@@Z)"
src\server_io_pylon.dll : fatal error LNK1120: ������������� ������� ���������: 1
```

The problem code:

```cpp
void AlarmOpcMapper::UpdateConditions(const AlarmInfo& alarm_info, DiscreteAlarmType::SPtr discrete_alarm) const {
    auto acked_state = boost::dynamic_pointer_cast<VariableNodeClass>(alarm_info.acked_state_node.lock());
    auto active_state = boost::dynamic_pointer_cast<VariableNodeClass>(alarm_info.active_state_node.lock());
    auto message = boost::dynamic_pointer_cast<VariableNodeClass>(alarm_info.message_node.lock());
    auto priority = boost::dynamic_pointer_cast<VariableNodeClass>(alarm_info.priority_node.lock());

    OpcUaDateTime current_time = OpcUaDateTime(boost::posix_time::microsec_clock::universal_time());
    auto make_opc_data_value  = [&](OpcUaVariant var, OpcUaDateTime timestamp) -> OpcUaDataValue {
        OpcUaDataValue value;
        value.serverTimestamp(current_time);
        value.statusCode(OpcUaStatusCode::Success);
        value.variant()->copyFrom(var);

        return value;
    };

    OpcUaDataValue val = make_opc_data_value(*discrete_alarm->activeState_Id(), discrete_alarm->activeState_TransitionTime()->get<OpcUaDateTime>());
    active_state->setValue(val);

    val = make_opc_data_value(*discrete_alarm->ackedState_Id(), discrete_alarm->ackedState_TransitionTime()->get<OpcUaDateTime>());
    acked_state->setValue(val);

    val = make_opc_data_value(*discrete_alarm->message(), discrete_alarm->activeState_TransitionTime()->get<OpcUaDateTime>());
    message->setValue(val);

    val = make_opc_data_value(*discrete_alarm->severity(), discrete_alarm->activeState_TransitionTime()->get<OpcUaDateTime>());
    priority->setValue(val);

}
```

The linker didn't manage to find `AttributeBase::setValue`  method with a **constant** reference. Neither our code does request the constant signature, neither the stack defines the signature. My competence in C++ is not  enough to explain this behavior.  Actually, we would have had a compilation error if we used
the signature which the stack didn't define. 

Any way we've worked it around by adding additional method in `AttributeBase::setValue`:

[patch.txt](https://github.com/ASNeG/OpcUaStack/files/2839561/patch.txt)

@huebl , I guess the problem is rather in our project than in the stack, but I report the bug hoping you have an idea that is really happening here.

Moreover I have an offer to refactor AttributeBase class and make all setter-methods with constant references. It'll be compatible with the current interface but more comfortable:

```cpp
  node->setValue(OpcUaDataValue(5.0));   // isn't working now
  // a user should do:
 
  OpcUaDataValue value(5.0);
  node->setValue(value); 
```

@huebl , what do you think?

Version 3.6.2
OS: Windows 10
Compiler: MS Visual Studio 15 2017 Win64

huebl, 2019-02-07:
For version 3.7, the AttributeBase class has already been redesigned. 

		bool setValueSync(const OpcUaDataValue& value);
		bool setValue(const OpcUaDataValue& value);
		
If you need the feature in version 3.6 you can make a sticky pick.

commit 0ad87163bdcb4eb579d82c333996610c90e43161
Author: Kai Huebl <kai@huebl-sgh.de> 2018-12-30 16:47:17
Committer: Kai Huebl <kai@huebl-sgh.de> 2018-12-30 
use const in setValueSync

huebl, 2019-02-07:
In version 4 we should introduce const in all setter and getter methods.

flipback, 2019-02-07:
@huebl , thx! I think the lack of the feature in 3.6 will motivate me to write the documentation quicker.  :smiley:

huebl, 2019-02-07:
I do not see the entire code. But I assume that acked_state is a const pointer. In this case, setValue can not be called because it should also be const. 

I'm not sure. The Windows message confuses me a little bit.

huebl, 2019-02-07:
I'm not sure if a lambda function works in the current implementation. The function returns a DataValue object. Currently, this object has no copy constructor and no assignment operator. We should change that in the next version.

flipback, 2019-02-07:
> I do not see the entire code. But I assume that acked_state is a const pointer. In this case, setValue can not be called because it should also be const.

If it had been so, our workaround wouldn't have fixed the error. Actually, this code used to be compiled without problem, until we updated the stack and port the project onto 64x platform.

flipback, 2019-02-07:
> I'm not sure if a lambda function works in the current implementation. The function returns a DataValue object. Currently, this object has no copy constructor and no assignment operator. We should change that in the next version.
> 
> Supplement: Was already implemented for version 4.1

I guess the copy constructor is not needed in this case. To be precise when a variable is  returned from a method (or lambda) as a result, usually no copping happens in C++11. The variable is *moved* to lvalue reference.

huebl, 2019-02-08:
>I guess the copy constructor is not needed in this case. To be precise when a variable is returned from a method (or lambda) as a result, usually no copping happens in C++11. The variable is _moved_ to lvalue reference.

Yes, that's right

closed 2019-03-13 by flipback