ASNeG/OpcUaStack #126: OpcUa Server keeps opening TCP connections

saukijan, 2019-03-05:
Hi @huebl , 

We have an application based your OPC UA Stack (it implements the `ApplicationIf()` and populates the address space) and we noticed after running the server for extensive amount of time that the OpcUaServer3 process keeps opening TCP connections for some reason. You can see it by running the following call in Linux: 

```bash 
watch -n1 lsof -p $(ps aux | grep "OpcUa" | grep -v grep | awk '{print $2}')
``` 

After some time you will see that the number of TCP connections keeps growing.  This leads in extreme case to TCP connection error **"Too many open files"** during address resolving. 

From what we see, it seems that the method `DiscoveryClientRegisteredServers::loop()` at 
`[ASNeG]/OpcUaStack/src/OpcUaStackClient/ApplicationUtility/DiscoveryClientRegisteredServers.cpp:156` is what is causing this problem.  

From running the debugger we can see `sessionService_->secureChannelState()` is always `SessionService::SecureChannelState::SCS_Disconnected`  and the state of `OpcUaStackCore::SecureChannel` is always `OpcUaStackCore::SecureChannel::State::S_Reconnecting`

It seems that `DiscoveryClientRegisteredServers::loop()` does not keep track of the open TCP connections it called for previously and simply tries to open a new connection without handling the old connections. 

Could you look into this and let us know if this is an actual problem with the stack? 

### Setup information:

| Commit ID | 6baaf83da5ad98eb2976211b56816cd7af54acbd |
| :-----------: | :-----------------------------------------------------: |
| **OS** | Ubuntu 16.04 |
| **Server configuration** | [ConfigurationFiles.zip](https://github.com/ASNeG/OpcUaStack/files/2931294/ConfigurationFiles.zip) |

huebl, 2019-03-05:
Thanks for the detailed error description. We will promptly analyze the error.

huebl, 2019-03-05:
**Reason**
The discovery process is repeated every 40 seconds. If successful, the OPC UA client opens an OPC UA connection to the discovery server, sends a discovery request to the discovery server and closes the connection. 
If a connection to the Discovery Server can not be established, the connection is not deleted and a new connection is opened after 40 seconds.

**Solution 1:**
The Discovery server is mostly not used. For this reason, the Discovery Server can be removed from the configuration. 

_/etc/OpcUaStack/ASNeG-Demo/OpcUaServer.xml_
```
<!--
  <DiscoveryServer>
      <DiscoveryUrl>opc.tcp://localhost:4840</DiscoveryUrl>
      <RegisterInterval>40000</RegisterInterval>
  </DiscoveryServer>
-->
```

**Solution 2:**
If the connection to the Discovery Server can not be opened, it will be deleted.  The bug should be fixed in version 3.6.3.

saukijan, 2019-03-05:
Thank you for a quick response, we already deactivated the Discovery Server in our case.

PR #127 (2019-03-14): Delete the discovery client connection on error.
Closes #126.
closed 2019-03-14 by huebl