ASNeG/OpcUaStack #130: Local build adds build_local_Debug to path

strowk, 2019-03-07:
While following 
https://opcuastack.readthedocs.io/en/latest/getting_started/hello_world.html
and running 
```
sh build.sh -t local -i hello-dist/
```
in directory `~/asneg-sample/helloworld` 
I expected the result to be in `~/asneg-sample/helloworld/hello-dist/` , but it was in
`~/asneg-sample/helloworld/build_local_Debug/hello-dist/`

flipback, 2019-03-07:
@strowk , never thought, someone would try to use relative path... 

I think we can change it. @huebl , can we?

huebl, 2019-03-07:
Hello @strowk, first of all many thanks for your contribution. We can change the installation script. We will change the installation script so that a relative path can be used. 

@flipback - I can take on this task. I would then create a fix for version 3.6.3.

flipback, 2019-03-07:
@huebl, no problems. You fix this bug, I'll fix the documentation.

closed 2019-03-19 by huebl