ASNeG/OpcUaStack #136: Get conditions events in Alarms tab of Event View pane

Phoenixreturn, 2019-03-13:
When I'm implementing these scheme to get alarms:
[image]
I can see these events:
[image]
but on alarms tab of Event View pane there is no event at the same time. What can be wrong?
[image]

I use:
OpcUaStack 
UA Expert 1.4.4
Windows 7

flipback, 2019-03-13:
@Phoenixreturn Without code, it's impossible to understand what you're doing wrong. Add some code snippet to help us, please.

huebl, 2019-03-13:
@Phoenixreturn 
If the log file for the OPC UA server still exists, please provide it. That would simplify the analysis of the problem. The file is in the directory <Install-Path>/var/log/OpcUaStack/ASNeG-Demo.

Thanks in advance.

Phoenixreturn, 2019-03-13:
@huebl , @flipback  
I've found the solution. The problem was in the following lines of code:
 ``variant->setValue(_alarm_object.alarm_condition_.getNodes()[0].lock().get()_);``
`` discrete_alarm->setAlarmConditionType(variant);``

You should pass _`alarm_object.alarm_condition_.getNodes()[0].lock()->getNodeId().get()`_
rather than what was before

huebl, 2019-03-14:
That's good news. Thanks for that.

closed 2019-03-14 by huebl