ASNeG/OpcUaStack #182: Stack overwrites status of processed HistroryReadRequest

flipback, 2019-04-17:
**Describe the bug**

My application uses some different statuses during processing HistoryRead requests.

```cpp
    std::vector<std::string> fb_attr_tags;
    boost::split( fb_attr_tags, full_attr_name,
                  [] (char c)
                      { return c == '.'; }
                );

    DataValueSerializer prototype;
    InterfaceSerializable::Vector container;

    try{
        if (attributes_tree.at(fb_attr_tags.front()).at(fb_attr_tags.back())->Load(start_point.dateTime(), end_point.dateTime(), prototype, container)) {
            data_value_array->resize(container.size());

            for_each(container.begin(), container.end(), [&](decltype(container)::value_type& value){
                auto data = constructSPtr<OpcUaDataValue>(std::dynamic_pointer_cast<DataValueSerializer>(value)->get_data_value());
                data_value_array->push_back(data);
            });

            if (container.size())
                return OpcUaStatusCode::Success;

            return OpcUaStatusCode::GoodNoData;
        }
    } catch (std::out_of_range& e) {
        LOG_WARN_FMT("Attribute %s is not historising : %s", full_attr_name.c_str(), e.what());
    }

    return OpcUaStatusCode::BadInvalidArgument;
```

But the stack overwrites them [here](https://github.com/ASNeG/OpcUaStack/blob/c7849a6a8cb6cffa317ee16f30e9044b2a84a141/src/OpcUaStackServer/ServiceSet/AttributeService.cpp#L577) by `InternalError`.

**Expected behavior**

An user application should have an ability to pass different statuses to a client. 

**Environment (please complete the following information):**
 - OS: Any
 - Version 3.7.0

flipback, 2019-04-17:
@huebl , I can fix it. Check, please, if it is a wrong behavior.

huebl, 2019-04-17:
Hi @flipback , I also think Bad InternalError is not the correct status code.

PR #265 (2019-07-10): Fix ReadHistory result status code
Closes #182

* **Please check if the PR fulfills these requirements**
- [x] The commit message follows our guidelines
- [x] Tests for the changes have been added
- [x] CHANGELOG have been updated

* **What kind of change does this PR introduce?**

Bug fix.

* **What is the current behavior?**

There is no way to pass the result status of ReadHistory service from the user app. It was overwritten by *InternalError*.
closed 2019-07-10 by huebl