ASNeG/OpcUaStack #189: Integrate json interface into read, write and historical read service

huebl, 2019-05-03:
The OPC UA Web Server is used to access an OPC UA server via a json Web socket interface. For this purpose, the following functions must be provided in json format.

- Read request/response
- Write request/response
- Historical Read request/response

flipback, 2019-05-03:
Hi, @huebl! Do you mean the server or client part? And how does it help OpcUaWebserver if it'll work with other OPC UA server?

huebl, 2019-05-03:
My description above was a bit confusing. I adapted this. 

The web application sends e.g. a read request in json format to the opc ua web server. The opc ua web server forwards the read request in the opc ua binary format to the OPC UA server. Both requests are identical. The format is json and the other times opc ua binary.

flipback, 2019-05-03:
@huebl, just to make sure I've understood you correctly... 

It's to optimize communication between OpcUaWebserver and OpcUaStack. If the web server has an OPC UA Server as a source that doesn't support JSON format, it uses the binary format. Is it right?

huebl, 2019-05-03:
The OPC UA WebServer transparently forwards all requests from the Web application to an OPC UA server. The web application sends a JSON packet over a web socket to the OPC UA web server. The OPC UA Web Server forwards this packet to the OPC UA server in OPC UA binary format. 

The JSON format between OPC UA Web Server and OPC UA Server is currently not supported.

flipback, 2019-05-03:
Ok, now I see.  Why do we need it in the stack? Maybe it'd be better to implement it in OpcUaWebServer project.

huebl, 2019-05-03:
There are two reasons for implementing the feature in the OPC UA stack.

- Most data types in the OPC UA stack support the OPC UA binary format and the JSON format.
- In the specification the JSON format is allowed. At the moment we support this only for PubSub. It does not always have to stay that way.

PR #199 (2019-05-06): 189 integrate json interface to attribute service
Closes #189
closed 2019-05-06 by huebl