ASNeG/OpcUaStack #302: OpcUaStackClient::Client crushes when remote server shuts down.

TYZzzzzzz, 2019-09-27:
**Description**
If the Client tries to connect to unreachable server - the Client crushes.
If a remote server, the Client already connected to, shuts down - the Client crushes.

According to one test, if a remote server shuts down, after the Client succesfully disconnects from it - the Client crushes anyway (possibly on destructor call).

**To Reproduce**
1. Start OpcUaServer.
2. Connect Client to server.
3. Stop the server.
4. The client will crush.

**Expected behavior**
Client should not crush.

**Screenshots**
[image]
[image]

**Logs**
[connect_to_unreachable_server.log](https://github.com/ASNeG/OpcUaStack/files/3661824/connect_to_unreachable_server.log)
[shut_down_connected_server.log](https://github.com/ASNeG/OpcUaStack/files/3661826/shut_down_connected_server.log)

**Environment (please complete the following information):**
- ОС Windows 7 x64 SP1, repeats on Windows 10 too
- boost 1.58
- OpcUaStack v. 3 from master branch
- MSVC compiler

**Additional**
My Client wrapper implementation:
[opc_ua_client_helper.cpp.txt](https://github.com/ASNeG/OpcUaStack/files/3661857/opc_ua_client_helper.cpp.txt)

huebl, 2019-09-28:
@TYZzzzzzz  - Do you also use version 3 for the client or only for the server?

huebl, 2019-09-28:
I did tests with the opc ua stack version 3. There was no crash.

huebl, 2019-09-28:
I did the following tests with an OPC UA client (version 3) on Linux.

Test 1:
1. Startup opc ua server
2. Startup opc ua client
3. Stop opc ua server
4. Start opc ua server

Test 2:
1. Startup opc ua client
2. Startup opc ua server

Result: The behavior of the OPC UA client and the OPC Ua server is correct.

huebl, 2019-09-28:
@TYZzzzzzz - I can not repeat the mistake you described.

Can you create a small OPC UA client program with which I can reproduce the error?

TYZzzzzzz, 2019-09-29:
I didn't perform tests on Linux, at this moment my target platform is Windows, but I'll try to test Linux version tomorrow

Before creating of this issue, I've tested with Google testing framework the minimal program like this
```cpp
TEST_F(OpcUaClientHelperTest, shouldWorksNormallyThenConnectingToUnreachableServer) {
    // init my client helper with concrete ip on SetUp phase
    // client is std::shared_ptr<OpcUaClientHelper>, I've attached implementation of this
    // helper to issue description as `opc_ua_client_helper.cpp.txt`

    // at this moment target server is unreachable
    client->Connect();
    std::this_thread::sleep_for(std::chrono::seconds{10});

    // without sleeping test finishes succesfully
    // but with `sleep` it crushes after 1-2 seconds with assertion error in
    // boost::shared_ptr on Debug build version
}
```

huebl, 2019-09-29:
For the analysis of the error a small test program would help me a lot. The test program should run on Linux and optionally on Windows and should be able to be started from the command line. Only if I can reproduce the mistake, it is possible for me to help you specifically.

TYZzzzzzz, 2019-09-30:
so, I performed test on Linux and got the same error

I prepared a minimal testing program: https://github.com/TYZzzzzzz/minimalClientTesting.git

I'm using an outer pre-builded dependencies - Boost and Stack, if it isn't usable for you, tomorrow I can expand it further

huebl, 2019-09-30:
@TYZzzzzzz - Thanks for your support.

At the session level and in the SecureChannel level, timers are used. Sessionless connections use the OpenSecureChannel timers. Connections with a session use the session level timers. If the timers are active in both planes, this leads to a crash.

**Workaround:**
Disable the timer of the SecureChannel level. The error should then no longer occur.

```
    void InitSessionServiceConfig() {
        // init default configs for session
        CryptoManager::SPtr cryptoManager = constructSPtr<CryptoManager>();
        sec_srvc_cfg_.secureChannelClient_->cryptoManager(cryptoManager);
        sec_srvc_cfg_.secureChannelClient_->connectTimeout(2400000);
        //sec_srvc_cfg_.secureChannelClient_->reconnectTimeout(1024);
        sec_srvc_cfg_.secureChannelClient_->securityMode(SecurityMode::SM_None);
        sec_srvc_cfg_.secureChannelClient_->securityPolicy(SecurityPolicy::SP_None);

        sec_srvc_cfg_.ioThreadName("Serverio Client " + ip_ + " session thread");
        sec_srvc_cfg_.mode_ = SessionService::Mode::M_SecureChannelAndSession;

        sec_srvc_cfg_.sessionServiceIf_ = new SessionServiceStub();

        sec_srvc_cfg_.session_->sessionName("Client " + ip_ + " session");
    }

```

The session timer is activated by default if a session exists (Default is 3 seconds).  The session timer can be set as follows.

```
   void InitSessionServiceConfig() {
        // init default configs for session
        CryptoManager::SPtr cryptoManager = constructSPtr<CryptoManager>();
        sec_srvc_cfg_.secureChannelClient_->cryptoManager(cryptoManager);
        sec_srvc_cfg_.secureChannelClient_->connectTimeout(2400000);
        //sec_srvc_cfg_.secureChannelClient_->reconnectTimeout(1024);
        sec_srvc_cfg_.session_->reconnectTimeout(1024);
        sec_srvc_cfg_.secureChannelClient_->securityMode(SecurityMode::SM_None);
        sec_srvc_cfg_.secureChannelClient_->securityPolicy(SecurityPolicy::SP_None);

        sec_srvc_cfg_.ioThreadName("Serverio Client " + ip_ + " session thread");
        sec_srvc_cfg_.mode_ = SessionService::Mode::M_SecureChannelAndSession;

        sec_srvc_cfg_.sessionServiceIf_ = new SessionServiceStub();

        sec_srvc_cfg_.session_->sessionName("Client " + ip_ + " session");
    }
```

**ToDo:**
In the session, the timer of the OpenSecureChanel level must be deactivated automatically if a session timer exists.

TYZzzzzzz, 2019-10-01:
@huebl Thanks for helping, the problem was fixed

PR #303 (2019-10-01): 302 client crash when remote shuts down  v3
Closes #302 

* **Please check if the PR fulfills these requirements**
- [ ] The commit message follows our guidelines
- [ ] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)
- [ ] CHANGELOG have been updated (for bug fixes / features / docs)

* **What kind of change does this PR introduce?** (Bug fix, feature, docs update, ...)

* **What is the current behavior?** (You can also link to an open issue here)

* **What is the new behavior (if this is a feature change)?**

* * [... cut]
closed 2019-10-01 by huebl