ASNeG/OpcUaStack #309: Hello World example for OPC UA Client

flipback, 2019-10-26:
**Is your feature request related to a problem? Please describe.**

There is no documentation for the OPC UA Client part of the stack. 

**Describe the solution you'd like**

We need a document in 'Getting started' part, that describes OPC UA conection establishment and data reading with ASNeG OPC UA Client.

wedvjin, 2020-04-21:
Hi @flipback ,
Reckon we could together tackle this?

So far, with *helloworld* server running if the client is run as:
```
$ OpcUaClient3 -command FindServer
```
getting:
```
DBG [7f15a1521740] init namespaces
client service for command FINDSERVER error: get client access object failed: Session=OpcUaClient
```
That is using built in client; how would you get started writing the client part of the application?

flipback, 2020-04-23:
Hi @wedvjin , FindServer service is not implemented on the server side by default. We planned to do it in a separeted project - https://github.com/ASNeG/OpcUaDiscoveryServer 

But the error message from the client looks strange anyway.  I'll try to look into the weekends.

wedvjin, 2020-04-24:
Hi @flipback ,
I would be very grateful if you could point me into the right direction of compiling the client application utilizing the stack.
So far I've got to based on the SyncReadFromServer.cpp trying to compile with:
```
$ g++ -Wall -I. -lboost_system Client.cpp -o test
```
But I get enormous [errors](https://pastebin.com/wbzJaCUy) so that is probably not the intended way.

wedvjin, 2020-04-28:
To update on my findings:
I have managed to compile the client app using cmake and appropriate makefiles.
Right now with helloworld server running and when trying to connect to it with a client getting assertion error in one of the core's methods:
```
DBG [7f1caf12c3c0] service set manager starts io thread: IOThreadName=<VBIClient>
DBG [7f1ca96d1700] start thread: ThreadId=<7f1ca96d1700>
INF [7f1ca96d1700] resolver complete
INF [7f1ca96d1700] connect secure channel to server: Address=<127.0.0.1>, Port=<8888>
INF [7f1ca96d1700] secure channel to server connected: Address=<127.0.0.1>, Port=<8888>
SimpleClient: /home/user/OpcUaStack/src/OpcUaStackCore/SecureChannel/SecureChannelBase.cpp:520: void OpcUaStackCore::SecureChannelBase::asyncWriteOpenSecureChannelRequest(OpcUaStackCore::SecureChannel*, OpcUaStackCore::OpenSecureChannelRequest&): Assertion `applicationCertificate().get() != nullptr' failed.
Aborted (core dumped)
```
I am going to continue my research, but in the meantime if you get any clues please let me know.
Many thanks

huebl, 2020-04-28:
Hi @wedvjin ,

The client needs a certificate and a crypto manager. 

**VBI Interface**
https://github.com/ASNeG/OpcUaStack/blob/master/tst/OpcUaStackClient/ValueBasedInterface/VBIAsyncReal_Attribute_t.cpp

        connectContext.applicationCertificate_ = applicationCertificate;
        connectContext.cryptoManager_ = cryptoManager;

**Session Service Interface**
https://github.com/ASNeG/OpcUaStack/blob/master/tst/OpcUaStackClient/ServiceSet/ServiceSetManagerAsyncReal_Attribute_t.cpp

	sessionServiceConfig.secureChannelClient_->applicationCertificate(applicationCertificate);
	sessionServiceConfig.secureChannelClient_->cryptoManager(cryptoManager);