ASNeG/OpcUaStack #310: OpcUaProjectBuilder3 generates a project that crashes

flipback, 2019-10-29:
**Describe the bug**

If I generate a project with command

```
OpcUaProjectBuilder3 testApp TestApp 4840
```

The application fails at the start with the error:

```
$ OpcUaServer3 ~/.ASNeG3/etc/OpcUaStack/testApp/OpcUaServer.xml 
OpcUaServer3: /home/flipback/projects/ASNeG/OpcUaStack3/src/OpcUaStackCore/SecureChannel/SecureChannelBase.cpp:520: void OpcUaStackCore::SecureChannelBase::asyncWriteOpenSecureChannelRequest(OpcUaStackCore::SecureChannel *, OpcUaStackCore::OpenSecureChannelRequest &): Assertion `applicationCertificate().get() != nullptr' failed.
```

**Expected behavior**

The application must not fail.

**Logs**
[OpcUaServer.log](https://github.com/ASNeG/OpcUaStack/files/3784052/OpcUaServer.log)

**Environment (please complete the following information):**
 - Linux x64
- Version: 3.8

**Additional context**

The application works if I delete <DiscoveryServer> section or change port of
<EndpointUrl> in OpcUaServer.xml file.

flipback, 2019-11-27:
@huebl, this issue needs your attention. As we also discussed, the sections `<EndpointUrl>` and `<ProductUri>` in OpcUaServer.xml file must not have Url with 0.0.0.0 address as it has in release 3.
But the reason for the issue is the usage of the same IP and port (4840) for the discovery server and the endpoint. My suggestions:
1. The server must indicate the conflict of the URLs and log a message about the error
2. The default address  for `<EndpointRUL>` and `<Discovery>` is @HOSTNAME@

huebl, 2019-11-27:
Yes, we can do it that way.

PR #323 (2019-12-02): Fix conflict DiscoveryServer URL and application EndpointURL
Closes #310

* **Please check if the PR fulfills these requirements**
- [x] The commit message follows our guidelines
- [x] CHANGELOG have been updated

* **What kind of change does this PR introduce?**

Bug fix

* **What is the new behavior (if this is a feature change)?**

1.~~The server checks if DiscoveryServer URL conflict with Application EndpointURL and exits with a message in a log.~~
2. By default, endpoint is @HOSTNAME@
3. By default, DiscoveryServer configuration is switched off
closed 2019-12-02 by huebl