ASNeG/OpcUaStack #317: OPC UA application is not avaliable by hostname

flipback, 2019-11-29:
**Describe the bug**

If I use `@HOSTNAME@` for EndpointUrl the application starts ok, but it isn't available by this endpoint. I steel forced to use IP address.

**To Reproduce**
Steps to reproduce the behavior:
1. Generate an application with OpcUaProjectBuilder3
2. Replace 0.0.0.0 by `@HOSTNAME@`
3. Run the app
4. Try to connect with UAExpert

**Expected behavior**

The client must connect to the application by URL `opc.tcp://hostname:port/`

**Logs**
[OpcUaServer.log](https://github.com/ASNeG/OpcUaStack/files/3903891/OpcUaServer.log)
[hostnametcp.pcapng.zip](https://github.com/ASNeG/OpcUaStack/files/3903893/hostnametcp.pcapng.zip)

**Environment (please complete the following information):**
 - OS: Ubuntu-19.10 
 - Version 3.8.0

**Additional context**

In the logs one can see that the server binds a socket with IP `127.0.1.1` but UAExpert is looking for IP 127.0.0.1.

flipback, 2019-11-29:
@huebl, why don't we bind  IP 0.0.0.0 for the hostname?

huebl, 2019-11-30:
The endpoint url must be a valid address (IP address or domain name), because it is used in certificate checking. The address 0.0.0.0 is not a valid address. In version 3 there is no certificate check. Therefore, we could make an exception for version 3.

In version 4, the following behavior was implemented:

https://github.com/ASNeG/OpcUaStack/issues/115

That would also be a solution for version 3.

flipback, 2019-11-30:
@huebl, I agree with you that 0.0.0.0 is not a valid IP for the endpoint. I mean not to use it for endpoint, but to bind the listening socket with 0.0.0.0 when the application has `@hostname@` in EndpointURL. Actually, it is what you do in #115.  

I can try to copy this solution into Release 3. Ok?

huebl, 2019-11-30:
Yes, you can do that.

PR #328 (2019-12-05): Use any address if endpoint url contains a domain name
Closes #317

* **Please check if the PR fulfills these requirements**
- [x] The commit message follows our guidelines
- [x] CHANGELOG have been updated

* **What kind of change does this PR introduce?** 

Bug fix

* **Other information**:

I copied commit from Release4
closed 2019-12-05 by huebl