ASNeG/OpcUaStack #319: CreateNodeInstance doesnt' return the created node

flipback, 2019-11-29:
**Is your feature request related to a problem?**

When we create an OPC Node with service `CreateNodeInstance` we must find it then in the Information Model by using `GetNodeReference` service to set the data value type, rank and etc.

**Describe the solution you'd like**

It'd be more comfortable to return the reference on the created node  `CreateNodeInstance` so that we have no need to find it again.

```cpp
CreateNodeInstance createNodeInstance(
  "DynamicVariable",                            // name
  NodeClass::Enum::EnumVariabl,                 // node class
  OpcUaNodeId(85),                              // parent node id (Objects)
  OpcUaNodeId("Dynamic", 1),                    // node id
  OpcUaLocalizedText("en", "DynamicVariable"),  // display name
  OpcUaQualifiedName("DynamicVariable", 1),     // browse name
  OpcUaNodeId(47),                              // reference type id
  OpcUaNodeId(62)                               // type node id
);

if (!createNodeInstance.query(&this->service())) {
  std::cout << "createNodeInstance response error" << std::endl;
  return;
}

auto node = createNodeInstance.nodeReference().lock();
```

huebl, 2019-11-30:
I think the function described above makes sense.  It makes it easier for the developer to use the interface. 

The following conditions should be considered.

1. The query function creates one or more nodes depending on the data type of the node. It must be possible to return more frequent references.
2. The root node reference can be determined without further information about a function.
3.  The children`s child node references must be identified via browse paths
4. For performance reasons, the new feature should be optional. The feature is not always needed and should be turned on and off by a function.

flipback, 2019-11-30:
Why don't we do different classes for each case.  For this issue, just one - `CreateAndReturnNodeInstance`?

huebl, 2019-11-30:
In future, the CreateNodeInstance, the CreateVariableInstance and the CreateObbjectInstance classes must be support multiple feature (The source code generators will need this features).  The use of a feature is optional. Furthermore, one or more features must be executed atomically. Therefore, the different features can not be distributed to multiple classes.

**Example**: We create an object with many nodes. The following features are optionally supported.

1. Set default value for one or more variables (various attributes possible)
2. Get reference for one or more nodes
3. Register one or more method callbacks
4. Register one or more (historical) write callbacks
5. Register one or more (historical) read calbacks
6. Register one or more monitored item start/stop callbacks
7 other...

huebl, 2019-11-30:
My proposition:
We create a new issue and use the new issue as a user story. For each additional subtask we create another issue and connect this activity with the user story. So we get a good overview of the necessary feature and the work to be done.

@flipback - What do you think about?

flipback, 2019-12-01:
> Example: We create an object with many nodes. The following features are optionally supported.

    1. Set default value for one or more variables (various attributes possible)
    2. Get the reference for one or more nodes
    3. Register one or more method callbacks
    4. Register one or more (historical) write callbacks
    5. Register one or more (historical) read callbacks
    6. Register one or more monitored item start/stop callbacks

I'm not sure if I've understood right, do you want to integrate all these features into one function? I think a user can do these tasks with the Application User API. It'd be more flexible.

> My proposition:
We create a new issue and use the new issue as a user story. For each additional subtask we create another issue and connect this activity with the user story. So we get a good overview of the necessary feature and the work to be done. Is there a better solution in github for this?

It would be great. Unfortunately, we have no subtask. So that we have it is only references.  But we can make a list of the subtasks in the main issue. I think it will work.

huebl, 2019-12-01:
I think we have to support both variants. 

There are some reasons why the node/object/variable class should support all necessary features.
1. Each operation requires an asynchronous operation. Using a lot of objects can create a performance problem.
2, The creation of certain objects must be atomic. If, for example, a file object is created in several steps, then it is in an undefined state from the point of view of a client over a certain period. Once an object is visible through the external interface, all feature must work.

flipback, 2019-12-03:
So far it is not clear to me. Could you write an example of the API?

PR #477 (2020-11-11): 319 create node instance does not return created node
Closes #319 

* **Please check if the PR fulfills these requirements**
- [ ] The commit message follows our guidelines
- [ ] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)
- [ ] CHANGELOG have been updated (for bug fixes / features / docs)

* **What kind of change does this PR introduce?** (Bug fix, feature, docs update, ...)

* **What is the current behavior?** (You can also link to an open issue here)

* **What is the new behavior (if this is a feature chang [... cut]
closed 2020-11-11 by huebl