ASNeG/OpcUaStack #322: DiscoveryClient crashes

flipback, 2019-12-01:
**Describe the bug**

DiscoveryClient crashes by the connection to an OPC UA server.

**To Reproduce**
Steps to reproduce the behavior:
1. Generate an OPC UA application with OpcUaProjectBuilder3
2. In section `<DiscoveryServer>`of **OpcUaServer.xml** file point the URL of existent and running
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Logs**
Provide application server's logs (default path /ANEG_INSALL_PREFIX/var/log/OpcUaStack/APP_NAME/OpcUaServer.log)

**Environment (please complete the following information):**
 - OS: [e.g. iOS]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

flipback, 2019-12-01:
Hi @huebl , I need a clew how to fix it.

huebl, 2019-12-01:
Stack Trace:

```
#0  0x00007ffff5101428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007ffff510302a in __GI_abort () at abort.c:89
#2  0x00007ffff50f9bd7 in __assert_fail_base (fmt=<optimized out>, 
    assertion=assertion@entry=0x7ffff64cf290 "applicationCertificate().get() != nullptr", 
    file=file@entry=0x7ffff64cf238 "/home/huebl/devel/OpcUaStack4/src/OpcUaStackCore/SecureChannel/SecureChannelBase.cpp", 
    line=line@entry=520, 
    function=function@entry=0x7ffff64cfaa0 <OpcUaStackCore::SecureChannelBase::asyncWriteOpenSecureChannelRequest(OpcUaStackCore::SecureChannel*, OpcUaStackCore::OpenSecureChannelRequest&)::__PRETTY_FUNCTION__> "void OpcUaStackCore::SecureChannelBase::asyncWriteOpenSecureChannelRequest(OpcUaStackCore::SecureChannel*, OpcUaStackCore::OpenSecureChannelRequest&)") at assert.c:92
#3  0x00007ffff50f9c82 in __GI___assert_fail (assertion=0x7ffff64cf290 "applicationCertificate().get() != nullptr", 
    file=0x7ffff64cf238 "/home/huebl/devel/OpcUaStack4/src/OpcUaStackCore/SecureChannel/SecureChannelBase.cpp", line=520, 
    function=0x7ffff64cfaa0 <OpcUaStackCore::SecureChannelBase::asyncWriteOpenSecureChannelRequest(OpcUaStackCore::SecureChannel*, OpcUaStackCore::OpenSecureChannelRequest&)::__PRETTY_FUNCTION__> "void OpcUaStackCore::SecureChannelBase::asyncWriteOpenSecureChannelRequest(OpcUaStackCore::SecureChannel*, OpcUaStackCore::OpenSecureChannelRequest&)") at assert.c:101
#4  0x00007ffff641937b in OpcUaStackCore::SecureChannelBase::asyncWriteOpenSecureChannelRequest (this=0x12bee20, secureChannel=
    0x7fffe8000b00, openSecureChannelRequest=...)
    at /home/huebl/devel/OpcUaStack4/src/OpcUaStackCore/SecureChannel/SecureChannelBase.cpp:520
#5  0x00007ffff64464ed in OpcUaStackCore::SecureChannelClient::handleRecvAcknowledge (this=0x12bee20, secureChannel=0x7fffe8000b00, 
    acknowledge=...) at /home/huebl/devel/OpcUaStack4/src/OpcUaStackCore/SecureChannel/SecureChannelClient.cpp:228
#6  0x00007ffff641784a in OpcUaStackCore::SecureChannelBase::handleReadAcknowledge (this=0x12bee20, error=..., bytes_transfered=20, 
    secureChannel=0x7fffe8000b00) at /home/huebl/devel/OpcUaStack4/src/OpcUaStackCore/SecureChannel/SecureChannelBase.cpp:335
#7  0x00007ffff6427c1d in boost::_mfi::mf3<void, OpcUaStackCore::SecureChannelBase, boost::system::error_code const&, unsigned long, OpcUaStackCore::SecureChannel*>::operator() (this=0x7ffff0b7ab28, p=0x12bee20, a1=..., a2=20, a3=0x7fffe8000b00)
    at /usr/include/boost/bind/mem_fn_template.hpp:393

```

huebl, 2019-12-01:
The opc ua discovery client wants to open a secure channel. For this the opc ua discovery client needs a certificate. This certificate does not exist.

PR #327 (2019-12-05): Fix discovery client
Closes #322

* **Please check if the PR fulfills these requirements**
- [x] The commit message follows our guidelines
- [x] Docs have been added / updated (for bug fixes / features)

* **What kind of change does this PR introduce?**

Bug fix.
closed 2019-12-05 by huebl