ASNeG/OpcUaStack #349: cannot open shared object file: No such file or directory (libOpcUaStackCore.so.3)

wedvjin, 2020-04-07:
**Describe the bug**
A clear and concise description of what the bug is.
Hello,
Downloaded and installed OpcUaStack into $HOME/.ASNeG as in docs. When trying to create a project by using OpcUaProjectBuilder3 in $HOME/.ASNeG/usr/bin/ getting the error 
```error while loading shared libraries: libOpcUaStackCore.so.3: cannot open shared object file: No such file or directory```
The required library is located in lib folder as in default install.
**To Reproduce**
Steps to reproduce the behavior:
1. Download and install OpcUaStack as in docs
2. Try running OpcUaProjectBuilder3
3. Observe error

**Expected behavior**
Create a project

**Logs**
No logs generated

**Environment (please complete the following information):**
 - OS: Kubuntu
 - Version 18.04

**Additional context**
Getting started with ASNeG OpcUaStack, many thanks for conducting such a great work!

wedvjin, 2020-04-07:
Set correct path as described:
```
$ export PATH=$PATH:$HOME/.ASNeG/usr/bin
$ export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:$HOME/.ASNeG/usr/lib/
```
before using OpcUaProjectBuilder3 to create a project.

closed 2020-04-07 by wedvjin

flipback, 2020-04-08:
@wedvjin , thank you for the report!