ASNeG/OpcUaStack #352: create self signed certificate error

wedvjin, 2020-04-09:
**Describe the bug**
Followed all the steps as in [Hello, World of OPC UA!](in https://opcuastack.readthedocs.io/en/latest/1_getting_started/hello_world.html) and when trying to launch the server getting errors:
```
error:220A4076:X509 V3 routines:a2i_GENERAL_NAME:bad ip address
error:22098080:X509 V3 routines:X509V3_EXT_nconf:error in extension
```
and from logs:
```
2020-Apr-09 12:54:52.104394 ERR [7f884c8cd740] create self signed certificate error : SSLError=<error:220A4076:X509 V3 routines:a2i_GENERAL_NAME:bad ip address>, SSLError=<error:22098080:X509 V3 routines:X509V3_EXT_nconf:error in extension>
2020-Apr-09 12:54:52.104494 ERR [7f884c8cd740] init application certificate error
```
Same occurs if the project is without any modifications described in Hello, World doc.

**To Reproduce**
Steps to reproduce the behavior:
1. Create project `$ OpcUaProjectBuilder3 helloworld "My First OPC UA Application" 8888`
2. Built it `$ sh build.sh -t local -i ~/helloworldinstalled -s ~/.ASNeG`
3. Add lib to $LD_LIBRARY_PATH `$ export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:~/helloworldinstalled/usr/lib`
4. Launch the server `$ OpcUaServer3 ./etc/OpcUaStack/helloworld/OpcUaServer.xml`
5. Observe errors

**Expected behavior**
Server started and accessible at `opc.tcp://127.0.0.1:8888` (localhost)

**Logs**
```
2020-Apr-09 12:54:52.056020 INF [7f884c8cd740] Start OpcUaServer
2020-Apr-09 12:54:52.056659 INF [7f884c8cd740]   OpcUaServer version      : 3.8.2
2020-Apr-09 12:54:52.056775 INF [7f884c8cd740]   Boost Library version    : .
2020-Apr-09 12:54:52.056866 INF [7f884c8cd740]   Open SSL Library version : 1.1.00
2020-Apr-09 12:54:52.056949 INF [7f884c8cd740]   Config Directory         : /home/user/helloinstalled/./etc/OpcUaStack/helloworld
2020-Apr-09 12:54:52.057335 INF [7f884c8cd740] Load Application Library helloworld
2020-Apr-09 12:54:52.057445 INF [7f884c8cd740]   Library Name             : helloworld
2020-Apr-09 12:54:52.057527 INF [7f884c8cd740]   Config File              : 
2020-Apr-09 12:54:52.058738 INF [7f884c8cd740]   Library Version          : 1.0.0
2020-Apr-09 12:54:52.058884 INF [7f884c8cd740] init opc ua core stack
2020-Apr-09 12:54:52.058984 INF [7f884c8cd740] init crypto
2020-Apr-09 12:54:52.061512 INF [7f884c8cd740] init certificate : Enable=<1>
2020-Apr-09 12:54:52.061621 INF [7f884c8cd740] create certificate directory : Directory=</home/user/helloinstalled/./etc/OpcUaStack/helloworld/pki/trusted/certs/>
2020-Apr-09 12:54:52.061876 INF [7f884c8cd740] create certificate directory : Directory=</home/user/helloinstalled/./etc/OpcUaStack/helloworld/pki/reject/certs/>
2020-Apr-09 12:54:52.062049 INF [7f884c8cd740] create certificate directory : Directory=</home/user/helloinstalled/./etc/OpcUaStack/helloworld/pki/trusted/crl/>
2020-Apr-09 12:54:52.062198 INF [7f884c8cd740] create certificate directory : Directory=</home/user/helloinstalled/./etc/OpcUaStack/helloworld/pki/issuers/certs/>
2020-Apr-09 12:54:52.062362 INF [7f884c8cd740] create certificate directory : Directory=</home/user/helloinstalled/./etc/OpcUaStack/helloworld/pki/issuers/crl/>
2020-Apr-09 12:54:52.062510 INF [7f884c8cd740] create certificate directory : Directory=</home/user/helloinstalled/./etc/OpcUaStack/helloworld/pki/own/certs>
2020-Apr-09 12:54:52.062672 INF [7f884c8cd740] create certificate directory : Directory=</home/user/helloinstalled/./etc/OpcUaStack/helloworld/pki/own/private>
2020-Apr-09 12:54:52.104394 ERR [7f884c8cd740] create self signed certificate error : SSLError=<error:220A4076:X509 V3 routines:a2i_GENERAL_NAME:bad ip address>, SSLError=<error:22098080:X509 V3 routines:X509V3_EXT_nconf:error in extension>
2020-Apr-09 12:54:52.104494 ERR [7f884c8cd740] init application certificate error
2020-Apr-09 12:54:52.104525 INF [7f884c8cd740] init opc ua server stack information model (structure)
2020-Apr-09 12:54:52.104570 INF [7f884c8cd740] init opc ua server stack information model (data)
2020-Apr-09 12:54:52.104596 INF [7f884c8cd740] init opc ua server stack service
2020-Apr-09 12:54:52.104621 INF [7f884c8cd740] init opc ua server stack session
2020-Apr-09 12:54:52.104646 INF [7f884c8cd740] init application
2020-Apr-09 12:54:52.104670 ERR [7f884c8cd740] shutdown server, because init server error
```

**Environment (please complete the following information):**
 - OS: kubuntu
 - Version: 18.04

**Additional context**
Help appreciated

flipback, 2020-04-09:
Hi @wedvjin , thank you for the report. Yep, this is a bug. You can work around it by change the IP address 0.0.0.0 to 127.0.0.1 in *OpcUaServer.xml*:

```xml
<IPAddress>0.0.0.0</IPAddress>
```

wedvjin, 2020-04-09:
Hi @flipback , many thanks for the prompt answer.
To clarify, had to set:
```
 <IPAddress>@HOSTNAME@</IPAddress>
```
to:
```
<IPAddress>127.0.0.1</IPAddress>
```

flipback, 2020-04-09:
Sorry mistyped, yes, you are right.

huebl, 2020-04-13:
A check is missing in the source code!.

@HOSTNAME@ is specified as the IP address in the configuration. However, the host name is not converted into an IP address. The test and conversion must be built in.

https://github.com/ASNeG/OpcUaStack/blob/master/src/OpcUaStackCore/Certificate/CertificateInfo.cpp#L172

huebl, 2020-06-05:
This parameter specified the ip addresses for which the opc ua server open ports. So it has to be a correct ip address. A ip address format check has been implemented.

closed 2020-06-08 by huebl