ASNeG/OpcUaStack #356: Hello, World is not read by client

wedvjin, 2020-04-10:
**Issue**
Made all modifications as described in [Hello, World of OPC UA!](https://opcuastack.readthedocs.io/en/latest/1_getting_started/hello_world.html), but the client still reads the Value of the GreetingString node as *Ehmm*. It is not changed/updated to *Hello, world!* in Library.cpp by the following lines:
```
GetNodeReference getNodeReference(OpcUaNodeId(222,1));
<...>
OpcUaDataValue dataValue(OpcUaString("Hello, world!"));
ptr->setValueSync(dataValue);
```
Am I missing something?

wedvjin, 2020-04-10:
Modified the References in ./etc/OpcUaStack/helloworld/HelloWorldNodeSet.xml from:
```
<Reference ReferenceType="Organizes" IsForward="false">ns=1;i=1</Reference>
```
to:
```
<Reference ReferenceType="HasComponent" IsForward="false">ns=1;i=1</Reference>
```
The value is updated to *Hello, World!* and observed by the client.
The tricky part is if the change is reverted the *Hello, World!* is displayed.

flipback, 2020-04-13:
Hello @wedvjin , the reference type doesn't have any influence on the reading. 

Could you explain how you read the data?

huebl, 2020-04-13:
Hi @wedvjin and @flipback 

The UaExpert (Version 1.5.1) supports this access. However, it is also possible that other clients do not support this access. Which client was used for access?.

Why can a client cause problems?
The Organizes Reference should be used only for Objects of the ObjectType FolderType. See OPC UA Specification Part 3 Chapter 5.5. Clients could check this rule.

The reference Organizes should be replaced by the reference HasComponent.

flipback, 2020-04-13:
Hi @huebl , I always learn something new from you!

But the problem is strange anyway because according to the documentation the `GreatString` is placed into the folder (type i=61):

```xml
      <UAObject NodeId="ns=1;i=1" BrowseName="1:HelloWorldFolder">
          <DisplayName>HelloWorldFolder</DisplayName>
          <Description>The folder of the greeting string</Description>
          <References>
              <Reference ReferenceType="Organizes" IsForward="false">i=85</Reference>
              <Reference ReferenceType="HasTypeDefinition">i=61</Reference>
          </References>
      </UAObject>

      <UAVariable NodeId="ns=1;i=222" BrowseName="1:GreetingString" DataType="i=12">
          <DisplayName>GreetingString</DisplayName>
          <Description>The greeting string</Description>
          <References>
              <Reference ReferenceType="HasTypeDefinition">i=63</Reference>
              <Reference ReferenceType="Organizes" IsForward="false">ns=1;i=1</Reference>
          </References>
              <Value>
              <uax:String>Ehmm</uax:String>
          </Value>
      </UAVariable>
```

huebl, 2020-04-13:
@flipback : Yes you are right.

wedvjin, 2020-04-14:
Hello @flipback and @huebl ,
I am using UaExpert 1.5.1 .
I did some more tests:
Changed the string in Library.cpp to *Hello!*, rebuilt the application in a new folder (changed the IPAddress to 127.0.0.1 in OpcUaServer.xml); UaExpert connected and trusted new certificate; observed that the client still reads *Hello, World!* (not even *Ehmm*)
Then, I changed the reference type from ```Organizes``` to ```HasComponent``` and the client still reads *Hello, World!*.
It looks like in retains the previous string and now I am unsure what is actually wrong as the string is not updated and it seems that the change of reference type was more likely coincidence to update the string.

huebl, 2020-04-14:
Hi @wedvjin,

The OPC UA Server does not remember any information when restarting. Therefore the string "Hello World!" is not visible after a restart. Did you reinstall the application and then restart it?.  You can check the date of the library libhelloworld.so in the folder <INTALL_DIR>/usr/lib/. 

After changing the reference in the configuration (<INSTALL_DIR>/etc/OpcUaStack/helloworld/HelloWorldNodeSet.xml), the server must also be restarted.

Background:
After starting the OPC UA server, the information model is built up. After the start, the string "Ehmm" is visible in the client (From Nodeset configuration). Then the library writes the new value "Hello!" into the variable. Now the string  "Hello!" is visible in the client,

Can you provide a log file of this process?. 
Log-File: <INSTALL_DIR>/var/log/OpcUaStack/helloworld/

wedvjin, 2020-04-14:
Hi @huebl ,
Many thanks for your reply it allowed me to look into the right direction:
So the installed application library <INSTALL_DIR>/usr/lib was defined for the previous application, and the server used defined path for it to find libhelloworld.so of the previous installation (I was doing it all in the same terminal session).
After I opened a new session and defined path's appropriately:
```
$ export PATH=$PATH:$HOME/.ASNeG/usr/bin
$ export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:$HOME/.ASNeG/usr/lib/
```
and
```
$ export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:$HOME/<INSTALL_DIR>/usr/lib/
```
ensuring that there is only present application in the library path -- it all works smoothly now.
My bad, sorry for that.
Thank you for Your support!

closed 2020-04-14 by wedvjin

huebl, 2020-04-14:
You're welcome