ASNeG/OpcUaStack #503: How to create a Subscription in user application using OPCUA Client Stack

upendar25, 2021-06-10:
Please provide example code or at least some steps on how to create a subscription in user application using client stack.

huebl, 2021-06-10:
Hi @upendar25 , there are several examples in the Git repository.

huebl, 2021-06-10:
For further questions please do not hesitate to contact us.

upendar25, 2021-06-10:
Hello,

Thank you for the quick response.

Where is the data of monitored items published by the server? Where is the callback function to process this data?

I want to print the data coming from the server through this SubPub service. Could you please point me in right direction? Thank you.

huebl, 2021-06-10:
https://github.com/ASNeG/OpcUaStack/blob/Release4/tst/OpcUaStackClient/ServiceSet/ServiceSetManagerSyncReal_MonitoredItem_t.cpp#L58

upendar25, 2021-06-10:
Thank you. I was using the 3.8.1 version and this code was not there. I will check it.

upendar25, 2021-06-15:
Hello,

Its working when I added while(1).. Is there any other way through implicit io threads?

huebl, 2021-06-15:
The unit test differs only slightly from a real application.  After creating a new monitored item, the executing main thread should not end but should wait in an event loop for commands to be entered (e.g. terminate program). The updates of the monitored item are served by independent threads from a thread pool.   

For example, the main thread waits in a mutex condition in an event loop. If the program is to be terminated, the main thread terminates the event loop and closes the monitored item and the OPC UA connection. The threads in the thread pool are then also terminated. If the main thread is ended, the program also ends.

upendar25, 2021-06-16:
Hello,

I am trying to create the subscription using ClientSubscription class. But there is no way to set the datachangeeventHandler to this class since ClientSubscriptionIF which handles the notifications has been set to Private without a setter method. Please clarify if I can use this class.

huebl, 2021-06-16:
Thanks for the information. The class is currently no longer used. We're going to remove the class from the repo. 

If the class remains in the repo, it would have to be revised.

upendar25, 2021-06-16:
Hello,

I am facing error when I have added two monitoreditems to a subscription. I have added the code snippet and error message. Please help.

auto monitoredItemCreateRequest = boost::make_shared<MonitoredItemCreateRequest>();
	monitoredItemCreateRequest->itemToMonitor().nodeId()->set("nodeID1",2);
	monitoredItemCreateRequest->requestedParameters().clientHandle(2250);
	monitoredItemCreateRequest->requestedParameters().samplingInterval(100);

	auto monitoredItemCreateRequest2 = boost::make_shared<MonitoredItemCreateRequest>();
	monitoredItemCreateRequest2->itemToMonitor().nodeId()->set("nodeID2",2);
	monitoredItemCreateRequest2->requestedParameters().clientHandle(2251);
	monitoredItemCreateRequest2->requestedParameters().samplingInterval(100);

	//monCreateReq->itemsToCreate()->resize(1);
	monCreateReq->itemsToCreate()->resize(2);
	monCreateReq->itemsToCreate()->set(0, monitoredItemCreateRequest);
	monCreateReq->itemsToCreate()->set(1, monitoredItemCreateRequest2);

error message: /usr/include/boost/smart_ptr/shared_ptr.hpp:734: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = OpcUaStackCore::MonitoredItemCreateResult; typename boost::detail::sp_member_access<T>::type = OpcUaStackCore::MonitoredItemCreateResult*]: Assertion `px != 0' failed.

huebl, 2021-06-16:
Please publish the stack trace.

upendar25, 2021-06-16:
Hey..

Sorry.. My mistake. My OPC Server was down and i didn't check.. There are many log messages.. I couldn't make out this..

upendar25, 2021-06-17:
Hello,

I have one more question..:)

I am running subscription in another thread with while(1) inside the subscription function. That is working fine. When I am trying to get the session and use it in CommandConnect, its not working. How can I write to server and read through Subscription at the same time?

Thank you.

huebl, 2021-06-17:
The OPC UA Attribute Service is required to write a variable (See opc ua specification part 4). The attribute service is determined from the session service. The Attribute Service can then be used to read and write variables.   

Example to write a variable:
https://github.com/ASNeG/OpcUaStack/blob/Release4/tst/OpcUaStackClient/ServiceSet/ServiceSetManagerSyncReal_Attribute_t.cpp#L87

Variables can be read or written in any thread. For example the while () thread can be used for this.

upendar25, 2021-06-18:
Hello,

The client is not being connected to ProSys Simulation server. Are there any particular settings in CryptoManager or Certificate I should do. I am using your test code to connect.

huebl, 2021-06-18:
With regard to encryption, certificates and authentication, there are many ways in which an OPC UA client can connect to an OPC UA server. Which settings does the OPC UA server use?.

upendar25, 2021-06-18:
Error Log: 

DBG [7f554b7fe700] async connect event: SessId=<1>
DBG [7f554b7fe700] set session service mode: SessId=<1>, SessServiceMode=<Normal>, CacheSize=<0>
DBG [7f554b7fe700] resolve host: Host=<127.0.0.1>, Port=<48010>
DBG [7f554b7fe700] change state in session service: SessionServiceName=<urn:127.0.0.1..:>, OldState=<Disconnected>, NewSate=<Connecting>
DBG [7f554b7fe700] adress resolv: Address=<127.0.0.1>
INF [7f554b7fe700] connect secure channel to server: ChannelId=<1-0>, Address=<127.0.0.1>, Port=<48010>
INF [7f554b7fe700] secure channel to server connected: ChannelId=<1-0>, Address=<127.0.0.1>, Port=<48010>
INF [7f554b7fe700] send hello request: ChannelId=<1-0>, Address=<127.0.0.1>, Port=<48010>
DBG [7f554b7fe700] handle write complete - Hello: ChannelId=<1-0>
INF [7f554b7fe700] receive acknowledge: ChannelId=<1-0>, Address=<127.0.0.1>, Port=<48010>
DBG [7f554b7fe700] handle write complete - OpenSecureChannelRequest: ChannelId=<1-0>
DBG [7f554b7fe700] create secure channel key: SecurityToken=<1>, ExpireTime=<2021-Jun-18 12:23:51.882475>
DBG [7f554b7fe700] session service handle connect: SessId=<1>
DBG [7f554b7fe700] session send CreateSessionRequest: SessId=<1>, RequestId=<1>, SessionName=<urn:127.0.0.1:Envipco:PLCOPC-Client>
DBG [7f554b7fe700] change state in session service: SessionServiceName=<urn:127.0.0.1:Envipco:PLCOPC-Client>, OldState=<Connecting>, NewSate=<CreateSession>
DBG [7f554b7fe700] handle write complete - MessageRequest: ChannelId=<1-1>
DBG [7f554b7fe700] create session response error; close secure channel: SessId=<1>, ResultStatus=<BadNonceInvalid>
DBG [7f554b7fe700] change state in session service: SessionServiceName=<urn:127.0.0.1:Envipco:PLCOPC-Client>, OldState=<CreateSession>, NewSate=<Error>
DBG [7f554b7fe700] handle write complete - CloseSecureChannelRequest: ChannelId=<1-1>
DBG [7f554b7fe700] opc ua secure channel read message header error; close channel: ChannelId=<1-1>, Message=<Operation canceled>
ERR [7f554b7fe700] opc ua secure channel close: Id=<1>, AsyncSend=<0>, CloseFlag=<0>
INF [7f554b7fe700] secure channel to server closed: ChannelId=<1-1>, Address=<127.0.0.1>, Port=<48010>
DBG [7f554b7fe700] start reconnect timer: SessId=<1>, Timeout=<5000>
DBG [7f554b7fe700] change state in session service: SessionServiceName=<urn:127.0.0.1:Envipco:PLCOPC-Client>, OldState=<Error>, NewSate=<Disconnected>

huebl, 2021-06-18:
This is an output of the OPC UA client. Which settings does the OPC UA server use?. 

A recording from wireshark is also helpful.

upendar25, 2021-06-18:
[image]

huebl, 2021-06-18:
Please create a wireshark dump of the configuration.

Has the client certificate been released on the server? 

What messages does the server issue?

upendar25, 2021-06-18:
I will create the wireshart dump and send it later. There are no certificates from Client added to Server. Server says session created in connection log. No explicit messages are available in the tool.

upendar25, 2021-06-19:
Hi.. Please find the wireshark data dump.

My understanding is that problem is with certificates. How does server get the client certificate? Just wondering if there is a problem with the server configuration.

[OPCUA_Error_Wireshark.csv](https://github.com/ASNeG/OpcUaStack/files/6680998/OPCUA_Error_Wireshark.csv)

huebl, 2021-06-19:
I need the binray from wireshark. Ideally only the OPC UA packages.

Then I can view the content of the OPC UA communication.  The result codes in the packages provide an indication of the error.

upendar25, 2021-06-19:
Hi,

I have attached the new file. I hope this is the correct file. if not, please let me the exact data type you need.
[OPCUA_Data.txt](https://github.com/ASNeG/OpcUaStack/files/6681102/OPCUA_Data.txt)

huebl, 2021-06-20:
This is not the right file. I need the pcap file from wireshark.

upendar25, 2021-06-20:
Hello,

[WireShark_Output.zip](https://github.com/ASNeG/OpcUaStack/files/6682287/WireShark_Output.zip)

I was using your test suites to connect to PROSYS Simulation Server. The only thing I changed is, SERVER_URI(from the main window of ProSys Simulator) and ServerCertificationname. 

And if possible,  Please provide test suite without Security Enabled. Did you test your stack client with Siemens PLC OPC Server? 

Thank you for the support.

huebl, 2021-06-20:
Your PolicyId in the client does not match the PolicyId from Server. Please set the correct PolicyId and User Token Type on server or on client. Both must match. 

[image]

[image]

You can set the PolicyId in the OPC UA Client. 
sessionServiceConfig.session_->authenticationAnonymous("correct-policy-id-from-server");

I have downloaded and tested the PROSYS OPC UA simulation server.  It worked immediately. Most likely you are using a changed server configuration.

huebl, 2021-06-20:
Information about the user identity token can be found in the OPC UA specification (Part 4, Part 6).

upendar25, 2021-06-20:
Thank you. I will change the code and try again. 

With regard to ProSys Simulation Server, I have changed only the serverCertificatename in CryptoManagerTest. Could you please point me to the code in the stack codebase you used to connect to the Simulation Server. I know its a bit of hassle for you but the help you are providing is much appreciated.  Thank you.

upendar25, 2021-06-20:
And is there tool you used to analyse the wireshark output..?

upendar25, 2021-06-20:
I see that wireshark itself provides the analysis too..

huebl, 2021-06-20:
PROSYS OPC UA Simulation server test:
- Use newest OPC UA version from https://github.com/ASNeG/OpcUaStack/tree/Release4. The PROSYS server has an error (The client nonce is not optional). The newest ASNeG Release 4 contains a workaround for this error.
- Use unittest https://github.com/ASNeG/OpcUaStack/blob/Release4/tst/OpcUaStackClient/ServiceSet/ServiceSetManagerSyncReal_Session_t.cpp. Set the correct endpoint.

upendar25, 2021-06-21:
Hi..

Its working with ProSys Simulation Server. But I am using without Security. I will check with security also. Thanks for the support.

huebl, 2021-06-23:
If there are no more questions I will close the issue. 

You can also use slack for further questions. 
https://asneg-team@slack.com

upendar25, 2021-06-23:
Hi,

Sure. I am still testing but you can close. I will write in slack if some
other problem arises.

Thank you.

Best Regards
Upendar

closed 2021-06-23 by huebl