ASNeG/OpcUaStack #513: SessionTimeout is not being set correctly

upendar25, 2021-07-19:
Hello,

I am trying to create a session after connecting to the OPC Server. I set the requestTimeout and sessionTimerout as 3600000 ms but the session is being closed after 30 s(Looks like default for Siemens PLC).

And Also, I was expecting callback function sessionchangeHandler but I didn't get any.

Just to summarize,  I need a single read/write session for faster operations.

Could you please help?

Regards,
Upendar

huebl, 2021-07-20:
**Request Timeout**
The value for the request timeout variable should be chosen significantly smaller. For example 3 seconds.

upendar25, 2021-07-20:
auto sessionStateUpdate = [this](SessionBase& session, SessionServiceStateId sessionState) {
			// FIXME: todo
			OPCLog(Debug, "Session Status:").parameter("session", sessionState);
			SessionService& sessionservice  = static_cast<SessionService&>(session);
			
			if((sessionState == SessionServiceStateId::Disconnected)) {
				OpcUaStatusCode statusCode = sessionservice.syncConnect();
		        if (statusCode != Success) {
				std::stringstream ss;
				ss << "Session Timeout error:"<< " StatusCode="<< OpcUaStatusCodeMap::shortString(statusCode);
				errorMessage(ss.str());
				} else {
					OPCLog(Debug, "Session Reconnected");
				}
			}
		};

the lambda function sessionStateUpdate is set to sessionchangeHandler.

huebl, 2021-07-21:
**Deadlock in  Example**
The source code above leads to a deadlock. A session is synchronized by a strand. This means that the session code may only be executed by one thread at the same time. In the example above, the session thread executes an update callback method and blocks the session module. A synchronous method (syncConnect) of the session is called inside the session update callback method. But the session is still blocked by the session thread.  It's a deadlock. Only a asynchronous method can be used in this situation. 

**Automatically Reconnect**
A reconnect is carried out automatically by the OPC UA client. To do this, the reconnect timeout must have a value greater than 0 (time in milliseconds to reconnect). 

**Reconnect Example**
https://github.com/ASNeG/OpcUaStack/blob/513_session_change_handler/tst/OpcUaStackClient/ServiceSet/Reconnect_Session_t.cpp_#L44

**Session Change Handler**
The session change handler works correctly.  
1. Start OPC UA Client
```
2021-Jul-21 11:00:41.986067 DBG [7f586cc32780] change state in session service: SessionServiceName=<>, OldState=<None>, NewSate=<Disconnected>
2021-Jul-21 11:00:41.986692 DBG [80cf0c29-df76-4677-9ae9-66e0d465e5cd_1] change state in session service: SessionServiceName=<urn:127.0.0.1:Company:Company-Client>, OldState=<Disconnected>, NewSate=<Connecting>
2021-Jul-21 11:00:41.988633 DBG [80cf0c29-df76-4677-9ae9-66e0d465e5cd_1] change state in session service: SessionServiceName=<urn:127.0.0.1:Company:Company-Client>, OldState=<Connecting>, NewSate=<CreateSession>
2021-Jul-21 11:00:41.990103 DBG [80cf0c29-df76-4677-9ae9-66e0d465e5cd_1] change state in session service: SessionServiceName=<urn:127.0.0.1:Company:Company-Client>, OldState=<CreateSession>, NewSate=<ActivateSession>
2021-Jul-21 11:00:41.991124 DBG [80cf0c29-df76-4677-9ae9-66e0d465e5cd_1] change state in session service: SessionServiceName=<urn:127.0.0.1:Company:Company-Client>, OldState=<ActivateSession>, NewSate=<Established>
```

2. Stop OPC UA Server
```
2021-Jul-21 11:04:35.364611 DBG [80cf0c29-df76-4677-9ae9-66e0d465e5cd_1] change state in session service: SessionServiceName=<urn:127.0.0.1:Company:Company-Client>, OldState=<Established>, NewSate=<Disconnected>
2021-Jul-21 11:04:36.356508 DBG [80cf0c29-df76-4677-9ae9-66e0d465e5cd_1] change state in session service: SessionServiceName=<urn:127.0.0.1:Company:Company-Client>, OldState=<Disconnected>, NewSate=<Connecting>
2021-Jul-21 11:04:36.357799 DBG [80cf0c29-df76-4677-9ae9-66e0d465e5cd_1] change state in session service: SessionServiceName=<urn:127.0.0.1:Company:Company-Client>, OldState=<Connecting>, NewSate=<Disconnected>
Try reconnect each 1000 milliseconds... 
2021-Jul-21 11:04:37.356428 DBG [80cf0c29-df76-4677-9ae9-66e0d465e5cd_1] change state in session service: SessionServiceName=<urn:127.0.0.1:Company:Company-Client>, OldState=<Disconnected>, NewSate=<Connecting>
2021-Jul-21 11:04:37.357685 DBG [80cf0c29-df76-4677-9ae9-66e0d465e5cd_1] change state in session service: SessionServiceName=<urn:127.0.0.1:Company:Company-Client>, OldState=<Connecting>, NewSate=<Disconnected>
2021-Jul-21 11:04:38.356394 DBG [80cf0c29-df76-4677-9ae9-66e0d465e5cd_1] change state in session service: SessionServiceName=<urn:127.0.0.1:Company:Company-Client>, OldState=<Disconnected>, NewSate=<Connecting>
2021-Jul-21 11:04:38.357588 DBG [80cf0c29-df76-4677-9ae9-66e0d465e5cd_1] change state in session service: SessionServiceName=<urn:127.0.0.1:Company:Company-Client>, OldState=<Connecting>, NewSate=<Disconnected>
```

3. Start OPC UA Server
```
2021-Jul-21 11:05:52.571397 DBG [282ecff7-a3e6-4e6c-9d8b-634345e5df8d_1] change state in session service: SessionServiceName=<urn:127.0.0.1:Company:Company-Client>, OldState=<Disconnected>, NewSate=<Connecting>
2021-Jul-21 11:05:52.575070 DBG [282ecff7-a3e6-4e6c-9d8b-634345e5df8d_1] change state in session service: SessionServiceName=<urn:127.0.0.1:Company:Company-Client>, OldState=<Connecting>, NewSate=<CreateSession>
2021-Jul-21 11:05:52.580041 DBG [282ecff7-a3e6-4e6c-9d8b-634345e5df8d_1] change state in session service: SessionServiceName=<urn:127.0.0.1:Company:Company-Client>, OldState=<CreateSession>, NewSate=<ActivateSession>
2021-Jul-21 11:05:52.581083 DBG [282ecff7-a3e6-4e6c-9d8b-634345e5df8d_1] change state in session service: SessionServiceName=<urn:127.0.0.1:Company:Company-Client>, OldState=<ActivateSession>, NewSate=<Established>
```

PR #514 (2021-07-28): add reconnect example
Closes #513

* **Please check if the PR fulfills these requirements**
- [ ] The commit message follows our guidelines
- [ ] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)
- [ ] CHANGELOG have been updated (for bug fixes / features / docs)

* **What kind of change does this PR introduce?** (Bug fix, feature, docs update, ...)

* **What is the current behavior?** (You can also link to an open issue here)

* **What is the new behavior (if this is a feature change)?**

* **Does this PR introduce [... cut]
closed 2021-07-28 by huebl